chipaudette/OpenAudio_ArduinoLibrary #4: Cannot get examples to compile in Arduino IDE.

incalcando, 2017-06-17:
I hope you don't mind me contacting you like this. I am not getting the OpenAudio examples to compile. They all fail with following error message. I have tested with an Teensy 3.6 with Audio Adapter and also with an Teensy 3.2 with Audio Adapter.

/Users/slexer/Documents/Arduino/libraries/OpenAudio_ArduinoLibrary/output_i2s_f32.cpp: In function 'float setI2SFreq(float)':
/Users/slexer/Documents/Arduino/libraries/OpenAudio_ArduinoLibrary/output_i2s_f32.cpp:44:176: error: narrowing conversion of '4.41176484e+4f' from 'float' to 'int' inside { } [-Wnarrowing]
   const int samplefreqs[numfreqs] = { 2000, 8000, 11025, 16000, 22050, 24000, 32000, 44100, 44117.64706 , 48000, 88200, 44117.64706 * 2, 96000, 176400, 44117.64706 * 4, 192000};
                                                                                                                                                                                ^
/Users/slexer/Documents/Arduino/libraries/OpenAudio_ArduinoLibrary/output_i2s_f32.cpp:44:176: error: narrowing conversion of '8.82352969e+4f' from 'float' to 'int' inside { } [-Wnarrowing]
/Users/slexer/Documents/Arduino/libraries/OpenAudio_ArduinoLibrary/output_i2s_f32.cpp:44:176: error: narrowing conversion of '1.76470594e+5f' from 'float' to 'int' inside { } [-Wnarrowing]

All examples form the 'original' Audio Library for the Teensy compile without any problems.

incalcando, 2017-06-17:
Sorry, I should add, Arduino 1.8.1, Teensiduino 1.36. On a Mac running OSX 10.12.4

chipaudette, 2017-07-06:
It's interesting that it actually *failed* to compile.  On my machine, it gives warnings on this variable down-sizing, but it compiles just fine.

But, whatever, I've fixed the code so that all of the examples compile on my machine without warnings or errors.  Hopefully it works for you!

Chip