chipaudette/OpenAudio_ArduinoLibrary #7: Teensy 4.0 compatibility

Orbital5k, 2019-10-05:
Not an issue as such - just wanted to say thank you for this library, I had a great time with the experimenting with it on the Teensy 3.6. Just wondering if the library will be updated for Teensy 4.0 compatibility? 

I was working on an effects processor on the 3.6 but in my code I get some errors that i'd love to help with if it's in my capabilities but not sure where to start and not sure of how big a change in core architecture the 4.0 has introduced. 

If I could help in any specific way just let me know and thanks again for the excellent library!

chipaudette, 2019-10-09:
Thanks for your interest!  Note that most of my Teensy floating-point audio development moved over to Tympan, which a device that is a Teensy 3.6 at heart with an attached audio codec.  Their library (here) is my library, plus a couple of years' worth of added development.  You might choose to give it a try!

With regard to your question here in my OpenAudio version of my library, I'll try to take a look tonight when I get home.

Chip

Orbital5k, 2019-10-10:
Hi Chip,

Thanks for getting back to me!

I will indeed check out the Tympan library, thank you. If you do get a chance to check my attempt at porting your compressor to the audio library that would be fantastic. I am not sure where I went wrong in the first attempt linked above but I am going to have another try over the weekend. 

Thanks again for making this library available and for your work on bringing easy floating point processing to the audio library!

Orbital5k, 2019-10-12:
Hi Chip,

I made another attempt to port the compressor to the standard audio library this time by bringing the audio_block_f32_t class into the compressor header file to minimise the amount of code that needed to be changed. I converted the incoming audio blocks to audio_block_f32_t in the header file but unfortunately this attempt has not been successful either (see attached).

Sorry for bothering you with this! I checked out the Tympan library and I am excited to see that you are testing a port to teensy 4!

I am a big fan of this compressor and used it a lot in my 3.6 projects so I will keep trying until I have it avaiable for the t4!

If you could point me in the right direction to porting the compressor that would be super helpful.

Thanks for any help!
[AudioEffectCompressor.h.zip](https://github.com/chipaudette/OpenAudio_ArduinoLibrary/files/3720617/AudioEffectCompressor.h.zip)

closed 2019-10-15 by Orbital5k