chipaudette/OpenAudio_ArduinoLibrary #9: Typos and tests

trash80, 2020-07-07:
Nice library!
Though I have found a typo in a critical part of the library unless I am mistaken...
https://github.com/chipaudette/OpenAudio_ArduinoLibrary/blob/master/AudioConvert_F32.h#L80
MAX_INT = 32678.0 // Shouldn't this be 32768? :D

Also at least on Teensy 4.0 with my bench testing the arm_q15_to_float and arm_float_to_q15 functions are a bit faster.

boblark, 2020-07-07:
Thanks Timothy.  No, in the conversion MAX_INT is a float!!  It is the 
number that a floating point number, scaled as -1.0 to 1.0 is multiplied 
by before being truncated to an integer.  It probably should have a "f" 
following the value, but the compiler will get it right anyway (the 
danger is that there will be a double to float conversion, but the 
compiler knows better).

Be aware, as listed at the start of the readme.md, that a lot of changes 
are happening to that library. One of these is in the area of 
input/output of codec data to the float library.  This should all be 
under control by July 1 :-)    Well, pretty soon anyway.  I am currently 
struggling with the I2S I/O including the conversion between I16 and 
F32.  It all works fine on T3 but not T4.

A lot of what is there is good right now, but some areas need work, like 
the T4 I/O.

Great to see you using the library.

Bob Larkin

trash80, 2020-07-07:
I get that, I am talking about the typo: 32**678** -> 32**768** 
Looking forward to the changes! Using this lib on a rather large project. :)

trash80, 2020-07-07:
Note: Unless that is done intentionally. It just seems like a typo, 67<->76 😉  Sorry if I am mistaken.

boblark, 2020-07-07:
Timothy -  Ooops, my dyslexia took over.  Thanks, I will correct that 
next push, probably within a day or so.  This is a great time to catch 
anything like that  as it is all in flux now.  Bob