chipaudette/OpenAudio_ArduinoLibrary #10: Example won't compile until "#include "DSP_Teensyduino_32.h" removed- then works fine

bmillier, 2020-09-14:
Great work on this F.P. audio library, Chip.
 I'm using Arduino 1.8.9 and Teensyduino 1.52. Until I remove the **#include "DSP_Teensyduino_32.h**" line from the TestEqualizer1 example, it won't compile. I don't see this file in the OpenAudio_Arduino library, so its not surprising. Compiles and works great without that line though. Am I missing something, or should the examples be edited to reflect this?
   DD4WH and I  have written floating point Convolution FIR filters using (FFT/iFFT) including segmented FFT routines that have very ow latencies even on FIR tap lengths up to 20K (for guitar cabinet simulation, for example). DD4WH did most of the original coding, and I ported it to a conventional PJRC audio library object and swapped in CMSIS routines for most all the processing. While the filter only accepts 16-bit stereo input/output, all the processing is in floating point, including of course the FIR coefficients.
   In the past, I had tried to implement a multiband EQ using our routine and  513 tap FIR coefficients. In theory, I figured I should be able to add the filter coefficients for each individual band together, but it worked poorly. When I came across your work, I could see that you also calculate the individual filter coefficients but use a window function, which I didn't.  When I use your equalizerNew routine to calculate the FIR coefficients, it looks like our filter is now working very nicely.
 I'm using Teeny 4.0 and 4.1, and it sounds like your openAudio library is not completely running on them yet. But, in the future I'd like to make use of your floating point Audio library further.   
Many thanks

chipaudette, 2021-01-05:
Bob, I don't now about TestEqualizer1.  It must be yours?  Can you address the DS_Teensuino_32.h comment?

boblark, 2021-01-06:
I believe that file is needed for some of the classes.  I will get it 
into the library for the places where it is needed.  It should get 
included into classes, not the INO.  I will check that.

On 1/5/21 1:02 PM, Chip Audette wrote:
> Bob, I don't now about TestEqualizer1. It must be yours? Can you address 
> the DSP_Teensuino_32.h comment?
>