chipaudette/OpenAudio_ArduinoLibrary #17: Bug fix for AudioEffectFormantShiftFD_OA_F32.h

DD4WH, 2022-09-27:
In lines 126, 128, and 132, please do not use N_2, but N_2 -1 
Otherwise the audio will contain nasty artefacts. 
N_2 is FFT_Size / 2 + 1, but the index should only have a max of FFT_Size / 2

Keep up the good work!

boblark, 2022-09-29:
Hi Frank - I saw that you were looking at frequency shifting of various sorts to help our ear's lack of high frequency response.  An interesting topic and a good one for Chip, I suspect.

On the bug fix, this class is really Chip's.  Chip, I can fix the constants as Frank  found?  Also, does this same issue show up in the Tympan library?  Thanks, Frank, for finding this.

Bob

DD4WH, 2022-09-29:
Hi Bob, 
in the meantime, I have modified the file(s) a little more in my trial to improve the audio quality. And I found a few more places with the same issue.
Maybe it is more straightforward, if I try to submit a pull request.
I will try to do that, but it will take me a few days / weeks more. :-) git is not my closest friend . . .