chipaudette/OpenAudio_ArduinoLibrary #18: No Audio in or out on Teensy4.0 with Rev D Audio board

TheCranston, 2022-10-04:
I've been trying to get any of the examples to produce audio to no avail.  That means "it's me" but I'm not sure what I'm doing wrong.  I have a Teensy 4.0 with the Dev D audio board.  The interface pins are different [https://www.pjrc.com/store/teensy3_audio.html](url) from the previous version, but I'm unsure off that's the issue.  The examples compile without error but no audio paths appear to be working or at least I cannot get audio out the headphone jack.  The stock PJRC audio library works so I know my card is functioning properly and the Arduino IDE is configured.

boblark, 2022-10-04:
Most of the examples run with LineIn rather than Mic.  This may be stopping the input.  The entire list of functions that can be used with the SGTL5000 is at http://www.janbob.com/electron/OpenAudio_Design_Tool/index.html?info=AudioControlSGTL5000   Bob

TheCranston, 2022-10-04:
Bob,
Thanks for the link.  I'm running LineIn and have headphones plugged into the headphones jack.  Is there some sort of special routing I need to get the audio to be present on the headphones jack?  I've loaded the PassthroughF32.ino example and can see the left channel peak bouncing around in the serial monitor, but alas no audio output.  thanks.