chipaudette/OpenAudio_ArduinoLibrary #19: SineCosOut.ino example not compiling for Teensy 3.2

psitech, 2023-01-15:
When trying to compile the SineCosOut.ino example for Teensy 3.2, compilation failed with the following error:
```
_C:\Users\Paul\Documents\Arduino\libraries\OpenAudio_ArduinoLibrary/AudioFilterConvolution_F32.h:107:31: fatal error: arm_const_structs.h: No such file or directory
compilation terminated._
```
Found an instance of arm_const_structs.h in _C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4\._ 
But this file is not present in _C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3_, hence the compilation error.

Environment: Windows 10, Arduino 1.8.19, Teensyduino 1.57.

Regards,
Paul

boblark, 2023-01-16:
Hi Paul - It would appear that the file|arm_const_structs.h |should be 
part of Teensyduino for teensy3 as it is for teensy4.  Or maybe there is 
some reason why it is omitted.  The Github source for that is 
https://github.com/PaulStoffregen/cores/tree/master/teensy3 and that 
includes the other two associated arm CMSIS math files.  teensy4 has all 
three so a compile on T4.x does not show the error.

Do you want to ask the Teensy forum about the missing file, or I would 
be happy to.

A temporary work around is to remove the convolutional files, cpp and h 
from OpenAudio and make a temporary change to the 
OpenAudio_ArduinoLibrary.h file to remove the reference  there.

Bob

psitech, 2023-01-16:
Hi Bob,
Thanks for your reply.
I did the temporary workaround as you suggested but then I ran into another compile error:
```
C:\Users\Paul\Documents\Arduino\libraries\OpenAudio_ArduinoLibrary/radioFT8Demodulator_F32.h:408:1: error: expected declaration before '}' token
 };    // End class def
 ^
```
Looking into radioFT8Demodulator_F32.h, the fourth line states: _*  Note: Teensy 4.x Only, 3.x not supported_ ...

Is it wise to just use a Teensy 4.x when playing with the OpenAudio_ArduinoLibrary?

Paul

boblark, 2023-01-16:
Hi Paul,  Yes, I had the brackets for the class define in the wrong 
place for radioFT8Demodulator_F32.h.  It was still trying to include 
that for T3.x which doesn't have adequate memory..  I fixed it and 
pushed it to github, so if you grab the radioFT8Demodulator_F32.h now 
and have it in place it should be OK.  Unless I am getting version 
confusion, which sometimes happens about now :-)

Also, I am currently looking at that convolution filtering class, and I 
see that it uses the new versions of CMSIS  FFT that are not allowed by 
the arm_math.h version in teensy3.  I would like to just preclude T3.x 
for that, but somebody may be already using it.  So I think we need to 
sort out the processors and change the FFT version to suit.  Another 
possibility is the kluge of borrowing the pieces of arm_math.h that are 
needed and use that with either T3.x or T4.x.  I'll try the latter as it 
is easier!

// Is it wise to just use a Teensy 4.x when playing with the 
OpenAudio_ArduinoLibrary?
I hope not!  The intention is to sort out the parts that are processor 
dependent and alter the code to suit.  I keep a T3.6 on my desk to try 
these out with, but, I sometimes forget!  Thanks for raising the 
question and the problems, as it gets these things looked at.  I tend to 
test things on T4.x most of the time.  But, we are getting more classes 
that benefit from the power of the T4.x. Each application has different 
needs, of course.

Bob

psitech, 2023-01-17:
Hi Bob, thanks for looking into this.
Are you sure you pushed radioFT8Demodulator_F32.h? 
You did change AudioFilterConvolution_F32.h in commit [a3ec83f](https://github.com/chipaudette/OpenAudio_ArduinoLibrary/commit/a3ec83fe5dc9148a3ac68151653435cd73e86772) but as far as I can see radioFT8Demodulator_F32.h was unchanged [and I did get the same compilation error as before after downloading a fresh copy of the library an hour ago].

I'm happy to hear that support for Teensy 3.x is continuing. But I also agree that Teensy 4 brings a lot more usable processing power.

Thanks,
Paul

psitech, 2023-01-20:
Hi Bob, thanks for correcting radioFT8Demodulator_F32.h. Found commit 2add4b4.
It now compiles fine so I can starting testing a sine generator using a Teensy 3.2 and a PCM5102A 32-bit, 384kHz DAC.

Paul

closed 2023-01-20 by psitech