chipaudette/OpenAudio_ArduinoLibrary #20: AudioSettings_F32 audio_settings, sample_rate_Hz fixed to 44117 Hz?

psitech, 2023-01-21:
While playing with the OpenAudio library on a Teensy 3.2 plus PCM5102A DAC to be used as a sine wave generator, I noticed that setting the _sample_rate_Hz_ did not have an effect. I tried 48000 Hz and 96000 Hz.
Here is the code used:
```
#include "OpenAudio_ArduinoLibrary.h"
#include "AudioStream_F32.h"
#include <Audio.h>

// T3.x supported sample rates: 2000, 8000, 11025, 16000, 22050, 24000, 32000, 44100, 44117, 48000,
//                             88200, 88235 (44117*2), 95680, 96000, 176400, 176470, 192000
// T4.x supports any sample rate the codec will handle.

const float sample_rate_Hz = 96000.0f;  // logic analyzer shows LRCK 44100 Hz, bit clock BCLK 2.82 MHz
const float freq = 1000.0f;
const int   audio_block_samples = 128;

AudioSettings_F32 audio_settings(sample_rate_Hz, audio_block_samples);

AudioSynthWaveformSine_F32 sine1;
AudioOutputI2S_F32         audioOutI2S1;
AudioConnection_F32        patchCord1(sine1, 0, audioOutI2S1, 0);
AudioConnection_F32        patchCord2(sine1, 0, audioOutI2S1, 1);

void setup() {
  AudioMemory_F32(20);
  sine1.amplitude(1.0);
  sine1.frequency(freq * 44117.647f / sample_rate_Hz);  // this shows ~460 Hz on oscilloscope, phase shifted
}

void loop() {
}
```

The logic analyzer shows this for the I2S signals:
[image]
As you can see the LRCK is ~44110 Hz. 
From the logic analyzer data I observed that audio data is 16 bits. How do increase that to 20 or even 24 bits?

I also noticed the sine output is phase shifted, although the mono sine source _AudioSynthWaveformSine_F32_ is just splitted to the L & R signals:
[image]
On the logic analyzer image you can also see that the left & right datablock is different within each LRCK sample.

Environment: Windows 10, Arduino 1.8.19, Teensyduino 1.57, Teensy 3.2, OpenAudio library up to and including commit https://github.com/chipaudette/OpenAudio_ArduinoLibrary/commit/88126e2b219e7f3f49bcfb3ce9721aba07d62f14.

Regards,
Paul

psitech, 2023-01-21:
OK, did some more digging through the code and examples.
The following code works for 48 kHz sample rate:
```
#include "OpenAudio_ArduinoLibrary.h"
#include "AudioStream_F32.h"
#include <Audio.h>

// T3.x supported sample rates: 2000, 8000, 11025, 16000, 22050, 24000, 32000, 44100, 44117, 48000,
//                             88200, 88235 (44117*2), 95680, 96000, 176400, 176470, 192000
// T4.x supports any sample rate the codec will handle.

const float sample_rate_Hz = 48000.0f; 
const float freq = 1000.0f;
const int   audio_block_samples = 128;

AudioSettings_F32 audio_settings(sample_rate_Hz, audio_block_samples);

AudioSynthWaveformSine_F32 sine1(audio_settings);                 // instead of sine1;
AudioOutputI2S_F32         audioOutI2S1(audio_settings);          // instead of audioOutI2S1;
AudioConnection_F32        patchCord1(sine1, 0, audioOutI2S1, 0);
AudioConnection_F32        patchCord2(sine1, 0, audioOutI2S1, 1);

void setup() {
  AudioMemory_F32(20);
  sine1.amplitude(1.0);
  sine1.frequency(freq);  // this gives the correct frequency, the next line does not
  //  sine1.frequency(freq * 44117.647f / sample_rate_Hz);  
}

void loop() {
}
```
No more phase shift as well:
[image]

When setting the frame rate to 96 kHz, the sine wave gets corrupted:
[image]
BCLK is now ~6.144 Mhz.

I guess the Teensy 3.2 is running out of steam...

Regards,
Paul

chipaudette, 2023-01-21:
I've only ever used this library on T3.6 and T4+, so I don't know if the clock-changing commands work for T3.2.  So, perhaps it's the T3.2 running out of steam...or more likely, it's that the commands used to adjust the I2S and DMA rates don't work for T3.2.

Sorry I'm not more help.

psitech, 2023-01-21:
Thanks Chip. Yeah, I'm already planning to move to Teensy 4 for performance reasons. It's just that have an PCB with Teensy 3.2 and the mentioned DAC mounted on it that I use a sine generator.
Perhaps Bob L can shed some more light on this, otherwise I will close this issue [I'm interested in 20- or 24-bit sine generation].

Paul

psitech, 2023-01-24:
I'm closing this issue [sample_rate_Hz fixed to 44117 Hz?] because it was actually a wrong invocation from my side.
These 2 lines should be invoked with (audio_settings) to override the default 44117 Hz.
```
AudioSynthWaveformSine_F32 sine1(audio_settings);                 // instead of sine1;
AudioOutputI2S_F32         audioOutI2S1(audio_settings);          // instead of audioOutI2S1;
```
Paul

closed 2023-01-24 by psitech