chipaudette/OpenAudio_ArduinoLibrary #21: AudioSynthWaveformSine_F32, stereo I2S output to DAC out-of-phase @ sample rate 44100 & 44117 Hz

psitech, 2023-01-22:
S/W: Windows 10, Arduino 1.8.19, Teensyduino 1.57, OpenAudio library up to and including commit https://github.com/chipaudette/OpenAudio_ArduinoLibrary/commit/88126e2b219e7f3f49bcfb3ce9721aba07d62f14. 
H/W:  Teensy 3.2 plus PCM5102A DAC.

While playing with the OpenAudio library generating a sine wave, I noticed that for sample rates 44100 & 44117 Hz, the sine output is out-of-phase.
Here is the code used:
```
#include "OpenAudio_ArduinoLibrary.h"
#include "AudioStream_F32.h"
#include <Audio.h>

// T3.x supported sample rates: 2000, 8000, 11025, 16000, 22050, 24000, 32000, 44100, 44117, 48000,
//                             88200, 88235 (44117*2), 95680, 96000, 176400, 176470, 192000
// T4.x supports any sample rate the codec will handle.

const float sample_rate_Hz = 32000,0f;
const float freq = 1000.0f;
const int   audio_block_samples = 128;

AudioSettings_F32 audio_settings(sample_rate_Hz, audio_block_samples);

AudioSynthWaveformSine_F32    sine1(audio_settings);           
AudioOutputI2S_F32            audioOutI2S1(audio_settings);         
AudioConnection_F32           patchCord1(sine1, 0, audioOutI2S1, 0);
AudioConnection_F32           patchCord2(sine1, 0, audioOutI2S1, 1);

void setup() {
  AudioMemory(20);
  AudioMemory_F32(20);
  sine1.amplitude(1.0);
  sine1.frequency(freq); 
  sine1.begin();
}

void loop() {
}
```

Output on scope at SR = 32000 Hz:
[image]

Output on scope at SR = 44100 Hz:
[image]

Output on scope at SR = 44117 Hz:
[image]

Output on scope at SR = 48000 Hz:
[image]

The logic analyzer on the I2S signals also shows different L&R samples at 44100 & 44117 Hz.

Paul

boblark, 2023-01-23:
> up to and including commit 88126e2
> SDS00073
> SDS00074
> SDS00075
> SDS00076

Paul, I assume the two traces are the left and right outputs.  I will 
try your code.

There is an issue that there is no certainty as to the output order from 
the SGTL5000 ADC.  It randomly starts up with an ambiguity as to which 
channel gets to I2S first.  But that is a tiny amount of time and 
nothing like what you are seeing.

Am I right that you are seeing a -1.0 multiplier on one of the 
channels?  Seems strange.

More later, Bob

psitech, 2023-01-23:
Hi Bob, thanks for looking into this.
> I assume the two traces are the left and right outputs 

Correct.
> Am I right that you are seeing a -1.0 multiplier on one of the channels?  Seems strange.

Yes, I fed the RCA cables from the [PCM5102A module](https://www.tinytronics.nl/shop/image/cache/data/product-1119/pcm5102-dac-audio-decoder-i2s-voorkant-600x600.jpg) directly into my scope.

If you want me to run some tests, just let me know.
Paul