chipaudette/OpenAudio_ArduinoLibrary #24: FFT sizes greater than 512 cause the Teensy to crash.

ReeceKetley, 2024-03-26:
When using the formant shifter setting sample size greater than 128 (256) which equates to 512 in the FFT size the Teensy crashes when trying to create the FFT object?

This is on a teensy 4.0 oveclocked to 712mhz

Is 512 the max I can get?

chipaudette, 2024-03-26:
Hi Reece,

Due to constraints on how the Teensy processes audio, you are forced to do an FFT (of whatever length) every time an audio block fills.  By default, the maximum audio block  length inherited from the Teensy Audio library is 128 points long (per channel).  If you use my Tympan FFT routines with an overlap of 75%, your maximum FFT size will indeed by 4*128 = 512 points.  So, you can't go longer than this without breaking the default that the maximum block size is only 128 points.

There are two solutions to you to get a longer FFT:

* If you need finer frequency resolution (hence a longer FFT) but you don't care about the higher audio frequencies, you can set your sample rate lower.  If you use half the sample rate, your 512 point FFT will now have twice the frequency resolution!

* Or, you can use my recent improvements in the Tympan_Library to use a block size that is longer than 128 points.  If you set the audio blocks to have a length of 256 points, it'll double your maximum FFT length.  Per [this Tympan Example](https://github.com/Tympan/Tympan_Library/blob/main/examples/02-Utility/LongAudioBlocks/LongAudioBlocks.ino) you can set a longer audio blocks simply by changing the line `const int audio_block_samples = 128` to a longer value like `const int audio_block_samples = 256`.
 
At some point, you will hit the longest FFT that is supported by the Teensy Audio Library (which is 4096?), so beware.

If you still have trouble, please post a link to your code and I can maybe help you.

Chip

ReeceKetley, 2024-03-26:
Hi Chip, 

Thanks for the quick response! Ultimately, I'm trying to increase the quality of the formant shifting, which after some tweaking I have managed to staying within the 128 point restriction. 

The formant shifter is to be used with only vocals, I was getting this weird aliasing, which I assumed was the FFT size. Anyway one of the solutions I came up with was to shift the frequency bands I'm processing (i.e. not from 20hz-20k) and only process where the vocals sit which has given me more bins which seems on the face of it to have made a difference.  

Whenever I tried to set audio_block_samples higher than 128 points, it would just flat out crash when trying to create the FFT. But your explanation has cleared that up! Tympan Library is something I never stumbled across, I may need to check that out!

chipaudette, 2024-03-26:
Oh, this is the OpenAudio library!  Sorry!  I got confused.  I thought that you were posting on the Tympan_Library.

I created OpenAudio library first and then I created the Tympan_Library started.  The OpenAudio library was then picked up and maintained by others (still in my GitHub account).  So the two libraries share a lot of DNA, but each has had new developments that are independent of each other.

It's all confusing...even to me!

boblark, 2024-10-10:
I was pawing through the issues and see this is open.  Chip, I sorta thought of the OpenAudio objects that you created, like formant shifter, as still being yours?   That said, I guess I have done trivial corrections to some of your stuff, as well. I don't see any problems with a fuzzy division of efforts.  But, it is still your  library and your call!!   Bob

chipaudette, 2024-10-10:
Bob, what kind of fix should be done for this?  It's a known limitation of all the frequency-domain examples that the max FFT is 4x the audio block length.  Since the max audio block size is 128, the max FFT size is 512.  It's inherent in the design and not an unexpected bug.

So, without re-engineering to remove the 4x limit (or the 128 block size limit), there's nothing to do to actually fix the problem.  I suppose that we could at least try to make it fail less hard?  What might you suggest?