chipaudette/OpenAudio_ArduinoLibrary #25: Compiler Warnings using Arduino IDE 2.3.2

Greg-R, 2024-08-24:
These are not show stoppers, but they are annoying.  Compiler output below.
Can any of these be resolved?

Thank you,
Greg

/home/raven/Arduino/libraries/OpenAudio_ArduinoLibrary/output_spdif3_f32.cpp: In member function 'void AudioOutputSPDIF3_F32::begin()':
/home/raven/Arduino/libraries/OpenAudio_ArduinoLibrary/output_spdif3_f32.cpp:77:15: warning: 'void* memset(void*, int, size_t)' clearing an object of type 'class audio_block_f32_t' with no trivial copy-assignment; use value-initialization instead [-Wclass-memaccess]
   77 |         memset(&block_silent, 0, sizeof(block_silent));
      |         ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/raven/Arduino/libraries/OpenAudio_ArduinoLibrary/output_spdif3_f32.h:27,
                 from /home/raven/Arduino/libraries/OpenAudio_ArduinoLibrary/output_spdif3_f32.cpp:28:
/home/raven/Arduino/libraries/OpenAudio_ArduinoLibrary/AudioStream_F32.h:34:7: note: 'class audio_block_f32_t' declared here
   34 | class audio_block_f32_t {
      |       ^~~~~~~~~~~~~~~~~
/home/raven/Arduino/libraries/OpenAudio_ArduinoLibrary/utility/BTNRH_rfft.cpp: In function 'int BTNRH_FFT::bitrev(int, int)':
/home/raven/Arduino/libraries/OpenAudio_ArduinoLibrary/utility/BTNRH_rfft.cpp:31:30: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
   31 |                 register int jj;
      |                              ^~
Opening Teensy Loader...
Memory Usage on Teensy 4.1:
  FLASH: code:211732, data:123428, headers:8896   free for files:7782408
   RAM1: variables:146240, code:191368, padding:5240   free for local variables:181440
   RAM2: variables:354496  free for malloc/new:169792

JayShoe, 2024-08-27:
SPDIF is also similar to optical/toslink. It is not the same as ADAT because ADAT has 8 channels and SPDIF has 2. I mention this since it sounds like you are just now learning about SPDIF. 

Firing one Teensy into the other is a great way to test it for sure. Get ADAT working and you are a hero.

boblark, 2024-08-27:
Thanks, guys.   I think it is a simple enough fix to be OK without hardware testing (I hate to say that, as it always ...).  We just need to zero the data with something like

for(int i=0;  i<block_silent.length;  i++)
	     block_silent.data[i] = 0.0f;

and get rid of the memset call.

I will try this, but it will be a couple of days.  Bob

boblark, 2024-10-10:
I think they are fixed.  Let me know of any problems.  Bob

Greg-R, 2024-10-10:
Compiled clean on T41 code and also the SPDIF output test rig.
Both are operating normally.  Looks good here, thank you!

I have an AudioOutputSPDIF3_F32 example sketch if you would like to add it.  Needs a little clean-up, and it will be ready to go.

boblark, 2024-10-10:
Thanks for the confirmation, Greg.  Yesterday I looked at your video on SPDIF test and ordered one of the SPDIF to DAC gadgets.  I was   planning to take your INO and make it simple as:  CODEC in and output to both CODEC and SPDIF.  I'll post it as an example, as you suggest.  Thanks for all the help.  Bob

Greg-R, 2024-10-11:
I have a question on SPDIF.  In the documentation, it says this:

"Native S/PDIF hardware is used, which is more efficient that use of I2S ports."

What exactly is the efficiency improvement?  Lower bandwidth, lower processor cycles, or what?
I'm looking at SPDIF as an alternative to I2S, especially if it saves processor cycles relative to I2S, which
is the current audio output interface.  Does anyone know the specifics behind this statement?