chipaudette/OpenAudio_ArduinoLibrary #26: TDM IO

Inconspicuous-person, 2024-12-26:
Could we please get the tdm i/o blocks implemented?

boblark, 2024-12-26:
I do not have any TDM hardware.  Is there someone that could test the software if we added the type conversions to the I16 code?

Meanwhile, it seems like it should work to put as many I16-to-F32 and F32-to-I16 conversion blocks as you have channels being used?  Please report if that works OK.   Bob

Inconspicuous-person, 2024-12-26:
I am currently waiting to get some tdm hardware. While yes I could use i16 to f32 some tdm codecs and the like support 24bit or even 32bit natively. It would be great if that could be implemented into the library (idk how with 24bit tho). I am going to have a cs42448 dev board soonish which can go up to 24bit for testing.

Inconspicuous-person, 2024-12-27:
Actually on further investigation the 24 bit inputs work by using the same 32 bit slot but just leaving the un-used bits set to 0. Also the outputs appear to be 32 bit.

h4yn0nnym0u5e, 2025-04-16:
PR #27 has conversion objects which should allow the full 24-bit resolution to be used with the standard TDM objects from the integer Audio library. The integer ports are in pairs, on with the MSword and the other with the LSword - connect to TDM ports in pairs.

boblark, 2025-04-18:
Yes, as you say.   I will put something together for a TDM object.  It looks like there would be 16 16-bit channels for the 256-bit total and we would take them pair by pair and produce 8 F32 outputs.  To my non-expert eyes, this would be the answer for either 24 or 32  fixed point inputs.  All the hard parts are already there in the I16 library.  

Along those same lines, I was told somewhere that I2S was stereo only, but NXP's I2S process obviously supports more, as Paul has x4, x6 and x8 in the I16 using extra I/O pins.  I was going to also get versions of at least x4 to avoid the conversion step.   Comments?  Anybody use x6 or x8?
Bob

h4yn0nnym0u5e, 2025-04-18:
With the PR #27 the existing "16x16-bit" TDM object needs no changes. It's usually used as an 8x32-bit one, with most codecs only getting the even-numbered ports connected because the 16-bit library doesn't emit the LSword. With the F32toI16x2 converter 8 bits of the LSword become usable. Some codecs can usefully be configured as 16-bit, though (see below).

For a proper F32 object you'd presumably want to have an 8-port object directly connectable from your F32 streams. It's probably a minor change from the existing code. There's also [my multi-TDM branch](https://github.com/h4yn0nnym0u5e/Audio/tree/feature/multi-TDM) which I'm hoping at some point  will get pulled in - I haven't done a PR as yet, I'd like a few more people to report it's OK, and it's a pretty niche use-case! But I think people who want a high channel count (>=8 each way) are probably also keen on >16-bit audio.

The >stereo I²S that Paul implemented is effectively just multiple stereo streams, sharing clocks. Each data stream is limited to 2 channels per pin. It's a reasonable way to get a moderate channel count with easily available hardware like the audio adaptor. TDM is "better" for channel counts over about 6 each way, but suffers from the fact that hardware isn't necessarily easy to come by. [This looks hopeful](https://forum.pjrc.com/index.php?threads/updated-8x8-and-16x16-audio.75569/); time will tell if he sells enough to keep it an off-the-shelf option.

palmerr23, 2025-05-27:
The standard TDM stream needs to have integer samples converted to F32 format.

I've got the basic 8x8 32bit TDM drivers working on my 8x8 Teensy Audio Board and the F32 conversions in place.

I've revising the 8x8 control object to properly handle 32 bit samples.

I'll tidy them up and post them on my github.