chipaudette/OpenAudio_ArduinoLibrary #30: AudioSynthWaveformSine_F32 end() function does not work.

Greg-R, 2025-05-07:
I'm using  AudioSynthWaveformSine_F32 to generate a sinusoidal data stream.
The data should cease by using the end() method, but it doesn't.  Here is a simple sketch.
The behavior is easily verified using a headphone.  The tones don't stop.

// Demonstration that sine.end() function does not actually end the data stream.
// Listen to headphones connected to Audio Adapter.

#include "OpenAudio_ArduinoLibrary.h"
#include "AudioStream_F32.h"
#include <Audio.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include <SerialFlash.h>

// GUItool: begin automatically generated code
AudioSynthWaveformSine_F32 sine2;          //xy=453,422
AudioSynthWaveformSine_F32 sine1;          //xy=456,339
AudioOutputI2S_F32       audioOutI2S1;   //xy=803,373
AudioConnection_F32          patchCord1(sine2, 0, audioOutI2S1, 1);
AudioConnection_F32          patchCord2(sine1, 0, audioOutI2S1, 0);
AudioControlSGTL5000     sgtl5000_1;     //xy=662,194
// GUItool: end automatically generated code

void setup() {
  AudioMemory(20);       //allocate Int16 audio data blocks
  AudioMemory_F32(20);   //allocate Float32 audio data blocks
  sine1.frequency(1000);
  sine2.frequency(440);
  sine1.begin();
  sine2.begin();
}

void loop() {
  // put your main code here, to run repeatedly:
  sine1.end();
  sine2.end();
}

I am using a switch to shut off the data so downstream objects don't use CPU cycles.  It seems trivial, but it is a big deal with
regards to performance of the radio receiver I am working on.

Regards,
Greg

boblark, 2025-05-07:
Hi Greg - I just left a fix at synth_sine_f32.cpp    I have not tested it but what could go wrong in 1 line :-)  Let me know if it is OK or I will test right away. Bob

Greg-R, 2025-05-07:
The change does make a difference.  But I still see a problem.
If I end() only one of the two sines, the output audio makes sense.  One channel's tone stops.
But if I end() both sines, then I get a raspy tone out of both sides of the earphone.  I expected silence.
I wrapped the calls to end() in AudioNoInterrupts()/AudioInterrupts().  This made no difference; the raspy output is still there.

boblark, 2025-05-07:
Interesting.  Do you have the flow diagram from the Design Tool?    Look for an alternative path, because when you say "return"  at the top of the update() it will stop all output audio blocks from coming out.  I think this is what we are doing.  This evening, I will try a really simple example, maybe modify one of the existing test examples.

Another quick test is to put a analyze_peak_f32 on the output of both sines.  When end() occurs it should not show data available.  If peak1.available() is ever true, I have something to fix, (I think).

boblark, 2025-05-08:
When I got up to speed, I realized you included the sketch that has the F32 objects.  So I have the flow diagram.   When you turn both sine waves off there is no exchange of data at all.  I will try that.  If it isn't silent it needs repair, I think!  Bob

Greg-R, 2025-05-08:
I screwed up that sketch.  I forgot to enable the device and turn on the headphones.
The following is the revised sketch.
Analyze_peak_f32 is added to both sines.  They return 1 when active, 0 when ended.  As expected.
But there is still that buzzy tone when both sines are ended.

// Demonstration after sine begin() and end() functions revised.
// Listen to headphones connected to Audio Adapter.
// Buzzy tone is emitted when both sines are ended with .end() method.

#include "OpenAudio_ArduinoLibrary.h"
#include "AudioStream_F32.h"
#include <Audio.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include <SerialFlash.h>

// GUItool: begin automatically generated code
#include "OpenAudio_ArduinoLibrary.h"
#include "AudioStream_F32.h"
#include <Audio.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include <SerialFlash.h>
#include "analyze_peak_f32.h"
#include "analyze_rms_f32.h"

// GUItool: begin automatically generated code
AudioSynthWaveformSine_F32 sine2;  //xy=453,422
AudioSynthWaveformSine_F32 sine1;  //xy=456,339
AudioOutputI2S_F32 audioOutI2S1;   //xy=803,373
AudioAnalyzePeak_F32 peak1;        //xy=806,274
AudioAnalyzePeak_F32 peak2;        //xy=818,443
AudioConnection_F32 patchCord1(sine2, 0, audioOutI2S1, 1);
AudioConnection_F32 patchCord2(sine2, peak2);
AudioConnection_F32 patchCord3(sine1, 0, audioOutI2S1, 0);
AudioConnection_F32 patchCord4(sine1, peak1);
AudioControlSGTL5000 sgtl5000_1;  //xy=662,194
// GUItool: end automatically generated code

void setup() {
  sgtl5000_1.enable();
  sgtl5000_1.volume(0.5);
  sgtl5000_1.unmuteHeadphone();
  AudioMemory(20);      //allocate Int16 audio data blocks
  AudioMemory_F32(20);  //allocate Float32 audio data blocks
  sine1.frequency(1000);
  sine2.frequency(440);
  sine1.begin();
  sine2.begin();
}

void loop() {

// Comment-uncomment the following 2 lines to deactivate the sines.
  sine1.end();
  sine2.end();

  Serial.printf("peak1.available = %d\n", peak1.available());
  Serial.printf("peak2.available = %d\n", peak2.available());

  delay(1000);
}

boblark, 2025-05-08:
Hi Greg - I have not gotten back to try this.  BUT, are you hearing the idle tone??   To the ear it sounds half like a tone and half noise.  Or some such.  Time was looking good, until this afternoon when the painters decided to come on Friday to power wash the house.  We need to get our junk out of the way and we don't work as fast as we used to!!  More as we see how this goes.  Bob

Greg-R, 2025-05-08:
Well this junk I am hearing is pretty loud.  Just guessing by ear it's the same magnitude as the sines.  Would an idle tone be below the level of the deterministic sines?

I will try the Teensy version of the same sketch and report.

Greg-R, 2025-05-08:
So the Teensy version of sine does not have an end() method.  So the only option is to set amplitude(0).
But you could also do that with the Open Audio sine, and it works, but it continues to consume processor cycles.

Now the next interesting thing, discovered by accident, if I run the Teensy version prior to the Open Audio version,
the nasty tone does not appear after setting end() on both sines!  BUT if I cycle the power with the Open Audio version
still loaded, it comes back up with the nasty tone.

So I wonder if there is some configuration of the SGTL5000 that is done by Teensy Audio, and because I didn't cycle the power, it was still in that state when running the Open Audio sketch?

Here is the Teensy sketch:

// Dual sinusoidal tones using Teensy Audio.

#include <Audio.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include <SerialFlash.h>

// GUItool: begin automatically generated code
AudioSynthWaveformSine   sine1;          //xy=474,324
AudioSynthWaveformSine   sine2;          //xy=480,407
AudioOutputI2S           i2s1;           //xy=948,335

AudioConnection          patchCord1(sine1, 0, i2s1, 0);
AudioConnection          patchCord2(sine2, 0, i2s1, 1);

AudioControlSGTL5000     sgtl5000_1;     //xy=638,175
// GUItool: end automatically generated code

void setup() {
  sgtl5000_1.enable();
  sgtl5000_1.volume(0.5);
  sgtl5000_1.unmuteHeadphone();
  AudioMemory(20);      //allocate Int16 audio data blocks
  sine1.frequency(1000);
  sine2.frequency(440);
}

void loop() {
// Deactivate sines.
  sine1.amplitude(0);
  sine2.amplitude(0);

  delay(1000);
}

Greg-R, 2025-05-08:
Another thing I discovered:  Increasing AudioMemory(70).  This makes the Open Audio sketch work as expected, 
in other words, the end() methods on the two sines completely eliminate anything coming out of the headphones, including any bizarre tones.

Seems like a lot of memory for this simple sketch.  But it works perfectly with active tones and AudioMemory(20)?

Greg-R, 2025-05-10:
I did more experiments this morning.  I found the 2 lines changed in the sinusoidal tone code.
So I could quickly move back and forth between the modified and unmodified code.

The improvement is massive!  The project I am working on (T41 radio) has a strong dependence on loop() processing time.
This one small change cuts the receive loop() processing time in half!  This is because the transmitter code, which uses 2 of the sine objects, was continuing to run, in spite of the end() method being used prior to entering receive mode.

Now I am still seeing some smaller anomalies in the loop() time.  But they are much smaller than caused by the non-function end() method, and are a lower priority.

This change is a huge improvement!  But it would still be good to know what causes the strange behavior in the simple sketch included above.

boblark, 2025-05-11:
Hi Gregg - I duplicated your results with the F32 and end() on both sine waves.   That tone, I think, is at the audio update rate, i.e., every 128 samples.

With end() on both, there is no longer any audio connection at all.  I am not sure what that would do, but the evidence is that it is not proper!  So, I added in the very least activity I could by outputting the number zero, every sample, always.  Like
AudioMathConstant_F32    mathConstant1;  //xy=205,423
AudioMixer4_F32               mixer4_1;       //xy=379,446 

AudioConnection_F32  patchCord11(mathConstant1, 0, mixer4_1, 0);
AudioConnection_F32  patchCord1(sine2, 0, mixer4_1, 1); 
AudioConnection_F32  patchCord12(mixer4_1, 0, audioOutI2S1, 0);

This ensures that there is always at least one connection.  It seems to stop the problem.  It requires almost no memory.

I know it is a requirement to have at least one input or output object.  Apparently in your case it requires slightly more with some minimal activity.   Most systems would provide that without anything special being done?

BTW, for an all F32 system, no I16 audio memory should be required.  We found a white noise generator that did and I am going to fix that.  We want no I16 memory to be needed.

boblark, 2025-05-12:
With the zero signals added on, the maximum F32 memory is 4 with the end() calls commented out, and is 3 with end() calls running.

Greg-R, 2025-05-12:
"I know it is a requirement to have at least one input or output object."

Well, that's a bit alarming.  I may have violated a basic rule.  Because my receiver audio code and transmitter audio code are independent!
So during normal operation, they operate one at a time, and the other is supposed to be completely shut off.

I took a look at your TinyTen diagram, and I see the transmitter and receiver are interwoven.

So is what I did with independent signal paths a no-no???

Greg-R, 2025-05-12:
So I was looking at this example recently.  Copied below, except for the loop();
Is this what you mean?  In this case there is a sort of floating AudioInputI2S.
I never would have noticed this before now.

Also, if you would have used the F32 I2S then you wouldn't need to allocate I16 memory?

/* AudioTestPeakRMS.ino  Bob Larkin 2 May 2020
 *  
 * Generates sine wave and measures RMS, Peak and Peak
 * to Peak values.
 * NOTE:  This is for the floating point _F32 versions of
 * AnalyzePeak and AnalyzeRMS *NOT* the Teensy Audio Library
 * versions with 16-bit Fixed point data blocks.
 */

#include "Audio.h"
#include <OpenAudio_ArduinoLibrary.h>
#include "analyze_peak_f32.h"
#include "analyze_rms_f32.h"

AudioInputI2S               i2s1; 
AudioSynthWaveformSine_F32  sine1;
AudioAnalyzeRMS_F32         rms1;
AudioAnalyzePeak_F32        peak1;
AudioConnection_F32         patchCord1(sine1, 0, rms1, 0);
AudioConnection_F32         patchCord2(sine1, 0, peak1, 0);

uint16_t n = 0;

void setup(void) {
  AudioMemory(5);       //allocate Int16 audio data blocks
  AudioMemory_F32(5);   //allocate Float32 audio data blocks
  Serial.begin(300);  delay(1000);

  // Default amlitude +/- 1.0
  sine1.frequency(1000.0);

  // Set next to 0 to suppress print errors in update()
  rms1.showError(1);
  peak1.showError(1);
}

boblark, 2025-05-12:
Hi Greg -  First, I think you should be alright with the "separate" T & R paths.  They are not really separate.  There is only one instance of the AudioStream_F32 stuff.  All memory comes from  one pool and all connections are controlled  as one.  The paths do not need to touch.  They all get serviced as needed every 128 samples via running the audio object function called update(), a special name.  The update() can receive 0 or more audio blocks corresponding to the number of input pins.  And it can send out 0 or more audio blocks corresponding to the number of output pins.  But that is figured out  by the update() and the Audio Stream just moves pointers around and the like.  The update() always gets run every 128 samples and nothing decides if a path is active or not, accept the update().

"Is this what you mean? In this case there is a sort of floating AudioInputI2S."   Yes, and that instantation of the I2S Input (or an Output would work as well) is necessary.  

"Also, if you would have used the F32 I2S then you wouldn't need to allocate I16 memory?"   That is correct.  It was somewhere around the time of that sketch when Chip wrote the F32 version of the I2S 2-channel I/O.  Now we do not need any I16 audio memory.

BTW, there are built-in functions for tracking memory usage (and CPU also).  See MyAudioEffect_Float.ino for an example.  I note that your 2 sine wave program uses about 1% of processor when running and 0.25% when not. Not very much.

Greg-R, 2025-05-12:
OK, all of my questions are answered on this subject.  The fix for the end() function for the sine object is verified working, and this is a significant improvement for my application.  Thank you!

I think we can close this one.

closed 2025-05-12 by Greg-R